ETLCPP/etl #1037: Add to_underlying To utility.h

drewr95, 2025-02-28:
I think it would be nice to add the `to_underlying` function that was added C++23.  I've implemented it locally on my C++14 codebase and figured it could be useful here.

```c++
#include <type_traits>

/**
 * @brief Converts an enum class value to it's underlying type'd value.
 *
 * @tparam Enum: Enum class type.
 * @param e: Enum class value.
 * @return constexpr auto
 */
template <typename Enum>
constexpr auto to_underlying(const Enum e) noexcept ->
    typename std::underlying_type_t<Enum>
{
    return static_cast<typename std::underlying_type_t<Enum>>(e);
}
```

jwellbelove, 2025-03-01:
An etl::underlying_type would need to be created to allow the use of no-STL platforms.

drewr95, 2025-03-10:
Related PR: https://github.com/ETLCPP/etl/pull/1045

rolandreichweinbmw, 2025-04-30:
etl::to_underlying is implemented in https://github.com/ETLCPP/etl/pull/1045 alongside etl::underlying_type and it doesn't depend on STL / std.

jwellbelove, 2025-05-22:
Version 20.41.0

closed 2025-05-22 by jwellbelove