ETLCPP/etl #1041: incorrect rounding of scaled natural numbers in scaled_rounding

dhebbeker, 2025-03-02:
Some of the scaled rounding functions incorrectly round scaled natural numbers. [For example scaled 5.0 is rounded to 6.0 with `round_infinity_scaled()`.](https://godbolt.org/z/c1z5EEhbj) According to the referenced paper in the documentation for [round-ceiling (toward positive infinity)](https://www.clivemaxfield.com/diycalculator/sp-round.shtml#A7) this should result to 5.0:

[image]

In the source code I see that before division with the scale, `SCALING` is added (or subtracted) to (or from) the rational parameter `value` in most rounding functions. Instead the term should be `(SCALING - 1)`. A fix would depend on the intended functionality:

- either adjust the documentation
- change the rounding in the source code

Can you please clarify which behaviour is intended? Then somebody could start a pull request...

<sub>Based on ETL version 20.40.0.</sub>

jwellbelove, 2025-03-02:
The functionality should follow the referenced paper.

jwellbelove, 2025-05-22:
Version 20.41.0