ETLCPP/etl #1042: incorrect rounding of in scaled_rounding with scale of 1

dhebbeker, 2025-03-02:
Most rounding functions in scaled_rounding don't work as intended if `SCALING==1`. I suggest to 

- either forbid to use a scaling of 1:
  - note in the documentation
  - `static_assert` in the code
- or use template meta-programming to return `value` if `SCALING==1`

Can you please clarify which behaviour is intended? Then somebody could start a pull request...

<sub>Based on ETL version 20.40.0.</sub>

jwellbelove, 2025-03-28:
It's a long time since I last looked at these.

jwellbelove, 2025-03-29:
As I look at this again after 7 years, I can't help thinking that scale factors that are not a power of 10 do not really make a lot of sense.

dhebbeker, 2025-03-30:
The `SCALING` could be the size of a container or frame. The following [example](https://godbolt.org/z/zn3xvojve) would make sense:

```c++
const unsigned numberOfNecessaryFrames = etl::round_infinity_unscaled<sizeof(Frame)>(payloadLength);
```

jwellbelove, 2025-03-30:
When I run that I get

`for 8 bytes and frames of size 8: 2 frame(s) are necessary`

Shouldn't that be 1 frame?

dhebbeker, 2025-03-30:
I agree. That is subject to issue #1041

jwellbelove, 2025-05-22:
Version 20.41.0

closed 2025-05-22 by jwellbelove