ETLCPP/etl #1059: CMake Deprecation Warning on 4.0.0

BartolomeyKant, 2025-04-01:
I'm always using an actual cmake version, currently it's 4.0.0 in my repository. Today I've added `etl` to my project for the first time and cmake generated warning

```
CMake Deprecation Warning at /home/bartolomey/projects/aether-client-cpp/third_party/etl/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.
```

Although it is not something terrible, but is it possible to rise up from `cmake_minimum_required(VERSION 3.5.0)` to at least `cmake_minimum_required(VERSION 3.10.0)`? 

P.S. Platformio on my machine uses cmake version 3.16.4 in its toolchain.

wsciaroni, 2025-05-14:
I believe this project should allow all non-deprecated versions of CMake to successfully configure.

Based on CMake's documentation, version 3.10 is now the oldest non-deprecated version as of the latest stable release.

My recommendation based on this is to transition this project's CMake up to 3.10.

wsciaroni, 2025-05-22:
@jwellbelove What version of CMake would you like to continue to support? The oldest non-deprecated version is 3.10.

PR #1066 (2025-04-30): rise up cmake minimum required version
Rise up to `cmake_minimum_required(VERSION 3.10)` as mentioned in [cmake warning message](https://github.com/ETLCPP/etl/issues/1059).

I don't know about other project but in my company we related to cmake minimum 3.16 as it is installed by default with platfomio. But it is also possible to use newer version as showed [here](https://github.com/platformio/platform-espressif32/issues/1538).

closes #1059
closed 2025-05-27 by jwellbelove