ETLCPP/etl #1064: v20.40.0 error: has no member named ‘is_secure’

KammutierSpule, 2025-04-02:
After updated my project to 20.40.0

I got this error:

```
error: ‘const class etl::ibasic_string<char>’ has no member named ‘is_secure’
  667 |         assign_impl(other.begin(), other.end(), other.is_truncated(), other.is_secure());
```

Looks to be related / missing a build option for
ETL_HAS_STRING_CLEAR_AFTER_USE

jwellbelove, 2025-04-03:
I'm not quite sure what may be happening there.

In `platform.h` there are these lines which should automatically enable `etl::basic_string::is_secure()` if `ETL_DISABLE_STRING_CLEAR_AFTER_USE` is not defined.

```
//*************************************
// Option to disable clear-after-use functionality for strings.
#if defined(ETL_DISABLE_STRING_CLEAR_AFTER_USE)
  #define ETL_HAS_STRING_CLEAR_AFTER_USE 0
#else
  #define ETL_HAS_STRING_CLEAR_AFTER_USE 1
#endif
```

KammutierSpule, 2025-04-03:
Sorry, what I mean is that the function

```
    void assign(const etl::ibasic_string<T>& other, size_type subposition, size_type sublength)
    {
      if (&other != this)
      {
        if (sublength == npos)
        {
          sublength = other.size() - subposition;
        }

        ETL_ASSERT(subposition <= other.size(), ETL_ERROR(string_out_of_bounds));

        assign_impl(other.begin() + subposition, other.begin() + subposition + sublength, other.is_truncated(), other.is_secure());
      }
    }
```

uses other.is_secure() even without check if ETL_HAS_STRING_CLEAR_AFTER_USE should be used or not.

and
```
#if ETL_HAS_STRING_CLEAR_AFTER_USE
...
    //*************************************************************************
    /// Gets the 'secure' state flag.
    //*************************************************************************
    bool is_secure() const
    {
      return flags.test<CLEAR_AFTER_USE>();
    }
#endif
```

So, it looks or the define (ETL_HAS_STRING_CLEAR_AFTER_USE) should be check on assign_impl(...) method call,
or the is_secure is always available:

```
    //*************************************************************************
    /// Gets the 'secure' state flag.
    //*************************************************************************
    bool is_secure() const
    {
#if ETL_HAS_STRING_CLEAR_AFTER_USE
      return flags.test<CLEAR_AFTER_USE>();
#else
     return false;
#endif
    }
```

KammutierSpule, 2025-04-03:
> In `platform.h` there are these lines which should automatically enable `etl::basic_string::is_secure()` if `ETL_DISABLE_STRING_CLEAR_AFTER_USE` is not defined.

but I have it defined :)
`#define ETL_DISABLE_STRING_CLEAR_AFTER_USE`

jwellbelove, 2025-04-03:
Maybe the functions should always exist. The 'unsecure' version would just return `false`.

jwellbelove, 2025-05-22:
Version 20.41.0

closed 2025-05-22 by jwellbelove