ETLCPP/etl #1071: etl::delegate

CodeBoy1213, 2025-04-10:
I am experiencing an issue with the ETL delegate when using lambda functions that capture variables. Below is the code snippet demonstrating the problem:

```
#include <cassert>
#include <iostream>
#include <array>

#include "etl/delegate.h"

class Foo
{
    public:
        Foo(int i): _i(i) 
        {            
            _printer = etl::delegate<void()>::create([&](){
                std::cout << _i << std::endl;        
            });
        }

        void Print()
        {
            _printer();
        }

    private:
        etl::delegate<void()> _printer;
        int _i;
};

int main() 
{
    std::array<Foo,3> arr = {{
        Foo(19),
        Foo(18),
        Foo(17)
    }};

    for(auto & i : arr)
    {
        i.Print();
    }
}
```
[Compiler Explorer](https://godbolt.org/z/j3149M3zY)

The delegate does not seem to work as expected. The captured variables do not go out of scope, only the lambda function does. I suspect the problem might be related to the lambda function not being copied, as described in the following issue:
[ETL issue #323](https://github.com/ETLCPP/etl/issues/323)

Is such an implementation with ETL possible?

jwellbelove, 2025-04-10:
The issue of supporting capturing lambdas and functors with member data is being discussed at the moment. There is the option of trying to modify `etl::delegate` or the prototype `etl::function` (std::function clone) or a new `etl::inplace_function` inspired by similar functionality in other libraries.

CodeBoy1213, 2025-04-11:
Thank you very much for the quick response.

Is it technically impossible at the moment for an etl::delegate to generally handle lambdas with captures? If so, is there a way to prevent the program from compiling when such lambdas are used?

CodeBoy1213, 2025-04-30:
Thank you for your response!

closed 2025-04-30 by CodeBoy1213

jwellbelove, 2025-04-30:
An update was pushed some time ago that added a deleted `create` function, so that temporary rvalue lambdas would cause a compile time error. I'm not sure why that didn't occur for your code.