ETLCPP/etl #1073: Test failure on 32-bit x86 arches

dkogan, 2025-04-23:
Hi. I'm seeing the build fail on i386 in Debian.

The failure is in test_gamma.cpp in test_int_gamma_encode. It expects result1a=(0, 0, 0, 1, 1, 2, 4, 5, 7, 9). Instead it gets (0, 0, 0, 0, 1, 2, 3, 5, 7, 9).

Here's why. Let's look at element 3. The logic is in gamma.h in `gamma_encode::operator()`: `TInput(TInput(maximum * pow(double(value) / maximum, one_over_gamma)));` where `TInput = int, maximum = 0.9, value = 3, one_over_gamma = 1./0.5`. Substituting everything in, we get `int(int(9 * pow(double(3) / 9, 1/.5)))`. This evaluates to `int(~1.0)`; depending on roundoff error, it can be `int(1-eps) = 0` or `int(1+eps) = 1`. This code is unstable. Is this intentional, and the test should be changed? Or should the code be fixed?

Thanks.

dkogan, 2025-04-23:
I removed the unstable values from the test: https://salsa.debian.org/debian/etlcpp/-/blob/master/debian/patches/fix-test-failure-gamma-i386.patch?ref_type=heads

Let me know if there's a better fix.

jwellbelove, 2025-05-22:
Version 20.41.0

closed 2025-05-22 by jwellbelove

dkogan, 2025-07-17:
Hi. Can I get a bit more info? I'm looking at 20.42.1, and I do not see this patch having been applied to that tree. Are you saying you merged this in 20.41.0, or you fixed the test failure in some other way?

Thanks