ETLCPP/etl #1085: Request: Add The Remaining Math Operators To ETL_TYPEDEF

drewr95, 2025-05-02:
I'm getting a clang build error of:

> error: conditional expression is ambiguous; 'const seconds' (aka 'const type_def<seconds_tag, float>') can be converted to 'float' and vice versa

When doing math on strong types.  I think all that's needed is the rest of the math operators need to be defined.  An example to reproduce the issue:
```c++
ETL_TYPEDEF(float, seconds);

bool useScalar{true};
constexpr seconds scalar{1.4F};
constexpr seconds defaultValue{1.0F};

seconds m_result = useScalar ? defaultValue : defaultValue * scalar;

// To actually get the above to compile you have to specify through temporary construction:
// seconds m_result = useScalar ? defaultValue : seconds{defaultValue * scalar};
```

jwellbelove, 2025-05-03:
I don't get any compiler errors for VS2022, clang 15.0.7, or GCC 12.3.0

Stepping through the code though, I could see a lot of conversions being made between the `etl::type_def` and integrals, which could probably be eliminated by more operators.

jwellbelove, 2025-05-03:
What C++ standard are you using?

drewr95, 2025-05-03:
I'm using Clang 20.1.1, and compiling for C++14 for my main source code.

jwellbelove, 2025-05-05:
Can you try this patch?

[etl-25-05-05-typedef.patch](https://github.com/user-attachments/files/20040744/etl-25-05-05-typedef.patch)

drewr95, 2025-05-08:
@jwellbelove That works great, thank you so much!  Now I'm eager for the next release :)

jwellbelove, 2025-05-22:
Version 20.41.0

closed 2025-05-22 by jwellbelove