ETLCPP/etl #1086: cyclic_value: Pointless Comparison Of Unsigned Integer With Zero

drewr95, 2025-05-05:
Using the Tasking compiler with C++14 and treating warnings as errors, I get the following compiler error:
`cptc E0186: ["thirdparty\etl\include\etl/cyclic_value.h" 116] pointless comparison of unsigned integer with zero`

This happens when 0 is the min or `First` value and in the interim I've had to use a signed type.

jwellbelove, 2025-05-20:
I've tried playing around with a bit of template mete-programming to see if I could resolve this, but with no success. 
According to the web, many compilers don't generate this warming in templated code due to this very issue.

drewr95, 2025-05-20:
Yeah, I'm extremely displeased with this compiler.  I may play around with it some to see if I could solve it.

Thank you very much for looking into this.

drewr95, 2025-05-20:
I was able to solve the compilation issue with the Tasking compiler.  Will you give this a try?  I'm not sure how much you're interested in this solution, and I understand if you don't desire to incorporate it.

[cyclic_value.patch](https://github.com/user-attachments/files/20350005/cyclic_value.patch)

jwellbelove, 2025-05-21:
If the type is `signed`, and `value < First`, and `First` is zero, then the check is not made.

I think that the truth table is this.

```
A           B         C 
First == 0  Type      value < First
No          unsigned  Check
Yes         unsigned  No check
No          signed    Check
Yes         signed    Check

C = !A + B
```

jwellbelove, 2025-05-21:
Does this work for you?

[etl-1086-cyclic_value-2.patch](https://github.com/user-attachments/files/20365168/etl-1086-cyclic_value-2.patch)

jwellbelove, 2025-05-22:
Version 20.41.0

closed 2025-05-22 by jwellbelove