ETLCPP/etl #1091: Clang20 Build Failure for C++17 and newer

wsciaroni, 2025-05-09:
I have seen a build failure due to a deprecation warning when using clang version 20. This is tested on the latest version of master 5896b3b90e401ec991ceda0097f8da3c8e9d8e9d

The build log is:

```log
--- Configuration Details ---
Compiler ID: clang20
C++ Standard (ETL_CXX_STANDARD): 17
NO_STL: OFF
ETL_FORCE_TEST_CPP03: OFF
CC: clang, CXX: clang++
--- Compiler Version ---
Debian clang version 20.1.4 (++20250426011826+182e8b7f8a71-1~exp1~20250426011849.108)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/lib/llvm-20/bin
--- CMake Version ---
cmake version 3.25.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
---------------------------
-- etl | Determined ETL version 20.40.0 from version.txt file
-- The CXX compiler identification is Clang 20.1.4
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiling for C++17
-- Compiling for STL
-- Using Clang compiler
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    ETL_FORCE_TEST_CPP03

-- Build files have been written to: /__w/etl/etl/build
[  0%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/AssertException.cpp.o
[  0%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/CompositeTestReporter.cpp.o
[  0%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/Checks.cpp.o
[  1%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/CurrentTest.cpp.o
[  1%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/DeferredTestReporter.cpp.o
[  1%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/DeferredTestResult.cpp.o
[  2%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/MemoryOutStream.cpp.o
[  2%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/ReportAssert.cpp.o
[  2%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/RequiredCheckException.cpp.o
[  3%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/RequiredCheckTestReporter.cpp.o
[  3%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/Test.cpp.o
[  3%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/TestDetails.cpp.o
[  4%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/TestList.cpp.o
[  4%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/TestReporter.cpp.o
[  4%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/TestReporterStdout.cpp.o
[  4%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/TestResults.cpp.o
[  5%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/TestRunner.cpp.o
[  5%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/ThrowingTestReporter.cpp.o
[  5%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/TimeConstraint.cpp.o
[  6%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/XmlTestReporter.cpp.o
[  6%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/Posix/SignalTranslator.cpp.o
[  6%] Building CXX object test/UnitTest++/CMakeFiles/UnitTestpp.dir/Posix/TimeHelpers.cpp.o
[  7%] Linking CXX static library libUnitTestpp.a
[  7%] Built target UnitTestpp
[  7%] Building CXX object test/CMakeFiles/etl_tests.dir/main.cpp.o
[  7%] Building CXX object test/CMakeFiles/etl_tests.dir/murmurhash3.cpp.o
[  8%] Building CXX object test/CMakeFiles/etl_tests.dir/test_algorithm.cpp.o
[  8%] Building CXX object test/CMakeFiles/etl_tests.dir/test_alignment.cpp.o
[  8%] Building CXX object test/CMakeFiles/etl_tests.dir/test_array.cpp.o
[  8%] Building CXX object test/CMakeFiles/etl_tests.dir/test_array_view.cpp.o
[  9%] Building CXX object test/CMakeFiles/etl_tests.dir/test_array_wrapper.cpp.o
[  9%] Building CXX object test/CMakeFiles/etl_tests.dir/test_atomic.cpp.o
In file included from /__w/etl/etl/test/test_algorithm.cpp:31:
In file included from /__w/etl/etl/test/../include/etl/algorithm.h:43:
In file included from /__w/etl/etl/test/../include/etl/iterator.h:[37](https://github.com/wsciaroni/etl/actions/runs/14938249081/job/41970566010#step:6:38):
In file included from /__w/etl/etl/test/../include/etl/private/addressof.h:[38](https://github.com/wsciaroni/etl/actions/runs/14938249081/job/41970566010#step:6:39):
In file included from /usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/12/memory:67:
/usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/12/bits/stl_tempbuf.h:263:8: error: 'get_temporary_buffer<TestDataNDC<int>>' is deprecated [-Werror,-Wdeprecated-declarations]
  263 |                 std::get_temporary_buffer<value_type>(_M_original_len));
      |                      ^
/usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/12/bits/stl_algo.h:4996:15: note: in instantiation of member function 'std::_Temporary_buffer<__gnu_cxx::__normal_iterator<TestDataNDC<int> *, std::vector<TestDataNDC<int>>>, TestDataNDC<int>>::_Temporary_buffer' requested here
 4996 |       _TmpBuf __buf(__first, (__last - __first + 1) / 2);
      |               ^
/usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/12/bits/stl_algo.h:5034:23: note: in instantiation of function template specialization 'std::__stable_sort<__gnu_cxx::__normal_iterator<TestDataNDC<int> *, std::vector<TestDataNDC<int>>>, __gnu_cxx::__ops::_Iter_less_iter>' requested here
 5034 |       _GLIBCXX_STD_A::__stable_sort(__first, __last,
      |                       ^
/__w/etl/etl/test/test_algorithm.cpp:1909:12: note: in instantiation of function template specialization 'std::stable_sort<__gnu_cxx::__normal_iterator<TestDataNDC<int> *, std::vector<TestDataNDC<int>>>>' requested here
 1909 |       std::stable_sort(data1.begin(), data1.end());
      |            ^
/usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/12/bits/stl_tempbuf.h:99:5: note: 'get_temporary_buffer<TestDataNDC<int>>' has been explicitly marked deprecated here
   99 |     _GLIBCXX17_DEPRECATED
      |     ^
/usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/x86_64-linux-gnu/c++/12/bits/c++config.h:119:34: note: expanded from macro '_GLIBCXX17_DEPRECATED'
  119 | # define _GLIBCXX17_DEPRECATED [[__deprecated__]]
      |                                  ^
[  9%] Building CXX object test/CMakeFiles/etl_tests.dir/test_base64_RFC2152_decoder.cpp.o
[ 10%] Building CXX object test/CMakeFiles/etl_tests.dir/test_base64_RFC2152_encoder.cpp.o
1 error generated.
gmake[2]: *** [test/CMakeFiles/etl_tests.dir/build.make:104: test/CMakeFiles/etl_tests.dir/test_algorithm.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:116: test/CMakeFiles/etl_tests.dir/all] Error 2
gmake: *** [Makefile:1[46](https://github.com/wsciaroni/etl/actions/runs/14938249081/job/41970566010#step:6:47): all] Error 2
```

jwellbelove, 2025-05-15:
The issue appears to be due to `std::get_temporary_buffer` being deprecated.
The ETL does not use or reference `std::get_temporary_buffer`.

wsciaroni, 2025-05-16:
> The issue appears to be due to `std::get_temporary_buffer` being deprecated. The ETL does not use or reference `std::get_temporary_buffer`.

Of course. I should have looked more closely.

My error was caused by using an invalid build environment. Closing this issue.