ETLCPP/etl #1092: Failure to Configure with CMake version less than 3.14

wsciaroni, 2025-05-10:
## Background

When attempting to do the CMake configure and using CMake version 3.13, the configure fails with the error:

This appears to be caused by this line in the top-level CMakeLists.txt:

```cmake
            # Generate the package configuration files using CMake provided macros
            write_basic_package_version_file(
                "${PROJECT_NAME}ConfigVersion.cmake"
                COMPATIBILITY SameMajorVersion
                ARCH_INDEPENDENT                                # This feature was added in CMake version 3.14
            )
```

## Potential Solution

Given that we want to preserve CMake compatibility with version 3.0.5 (based on the `cmake_minimum_required(VERSION 3.5.0)`) we could implement the following in place of the chunk above:

```cmake
        if(CMAKE_VERSION VERSION_GREATER_EQUAL 3.14)
            # Generate the package configuration files using CMake provided macros
            write_basic_package_version_file(
                "${PROJECT_NAME}ConfigVersion.cmake"
                COMPATIBILITY SameMajorVersion
                ARCH_INDEPENDENT
            )
        else()
            # This is needed for CMake < 3.14
            # because the ARCH_INDEPENDENT option is not available
            # in CMake < 3.14
            # See https://cmake.org/cmake/help/latest/module/CMakePackageConfigHelpers.html
            # for more information
            write_basic_package_version_file(
                "${PROJECT_NAME}ConfigVersion.cmake"
                COMPATIBILITY SameMajorVersion
            )
        endif()
```

BartolomeyKant, 2025-05-13:
I'm just wondering Why do you want to preserve CMake compatibility with version 3.0.5?

wsciaroni, 2025-05-14:
I think it's good for a library like this to support as wide a variety of build tool versions as reasonably possible.

That being said, I'm not tied to a specific version. However, I view updating the 'cmake_minimum_required' as a larger decision that should likely have its own issue and discussion.

BartolomeyKant, 2025-05-14:
> I view updating the 'cmake_minimum_required' as a larger decision that should likely have its own issue and discussion.

Like this https://github.com/ETLCPP/etl/issues/1059 ?

wsciaroni, 2025-05-14:
> > I view updating the 'cmake_minimum_required' as a larger decision that should likely have its own issue and discussion.
> 
> Like this https://github.com/ETLCPP/etl/issues/1059 ?

Yes... Exactly like that :)

Oops, I should have seen that one.

That being said, I found this error with CMake 3.13, so I believe it will still be valid if cmake_minimum_required is set at 3.10.

BartolomeyKant, 2025-05-14:
> as reasonably possible

What's why a asked the question!
Why it should be 3.0.5 or 3.10? Maybe if it uses features from 3.14 it should be cmake_minimum_required to 3.14.

wsciaroni, 2025-05-14:
> > as reasonably possible
> 
> What's why a asked the question!
> Why it should be 3.0.5 or 3.10? Maybe if it uses features from 3.14 it should be cmake_minimum_required to 3.14.

Let's move the conversation over to that issue in case others have input. :)

We can handle this issue based on the outcome of #1059

BartolomeyKant, 2025-05-14:
> Let's move the conversation over to that issue in case others have input. :)"

No. That issue has been move to Done and has to be closed, I suppose. @jwellbelove?

jwellbelove, 2025-05-15:
I've changed it back to 'in progress'.

PR #1098 (2025-05-22): Restore CMake < 3.14 compatibility
Closes #1092