ETLCPP/etl #1097: `etl::string::push_back('\0')` causes unintended behavior

Cpt-Seablue, 2025-05-15:
While messing around with unrelated logger shenanigans, I discovered that you can push a NULL terminator to a string. This effectively causes a silent truncation of your string because any later appends get lost in code that parses strings through the NULL terminator.

Is this intended behavior? Obviously it's quite stupid to push a NULL character to the string but we stumbled onto this issue when interfacing with a logger that pushed strings character by character, including NULL terminators.

To illustrate:
```C++
etl::string<32> t = "Hello";
t.push_back('\0');
t.push_back(' '); // Space character
t.append("World");

printf("%s", t.c_str()); // Prints "hello" not "helloA"
```

jwellbelove, 2025-05-15:
This is the expected behaviour.
`std::string` produces the same output.

Cpt-Seablue, 2025-05-15:
Ah, I see, thanks for the swift response.

closed 2025-05-15 by Cpt-Seablue