ETLCPP/etl #1101: Question\Feedback: does etl::array supports a size of 0?

vbaderks, 2025-05-18:
The std::array definition supports officially an array size of 0:

While solving some GCC warnings -Warray-bounds I noticed that the etl::array implementation uses the following implementation for the data() function:

  pointer data() ETL_NOEXCEPT
    {
      return &_buffer[0];
    }

https://github.com/ETLCPP/etl/blob/5896b3b90e401ec991ceda0097f8da3c8e9d8e9d/include/etl/array.h#L205

While the GCC compiler has an extension to allow usage of &_buffer[0], not all compilers do and some consider it invalid. 

It was not clear from the ETL documentation if using a size of zero is supported by the etl::array implementation or not.

jwellbelove, 2025-05-18:
I've just done a mod to allow zero sized `etl::array`.
It will be in the next release (very soon)

vbaderks, 2025-05-19:
@jwellbelove . Thanks for the quick response.

jwellbelove, 2025-05-22:
Version 20.41.0

closed 2025-05-22 by jwellbelove