ETLCPP/etl #1105: Multiple definition error etl::set_assert_function

TheComet, 2025-05-23:
```
/usr/bin/ld: libInitialisation.a(Initialisation.cpp.o): in function `etl::set_assert_function(void (*)(etl::exception const&))':
Initialisation.cpp:286: multiple definition of `etl::set_assert_function(void (*)(etl::exception const&))'; 
FlowmeterConfigurator.cpp:286: first defined here
collect2: error: ld returned 1 exit status
```

The steps to reproduce are:
 1. Compile with ```-DETL_USE_ASSERT_FUNCTION```
 2. In ```Initialisation.cpp``` I call ```etl::set_assert_function(EtlAssertCallback)``` once.
 3. In ```FlowmeterConfigurator.cpp``` I add the line ```#include <etl/vector.cpp>```

The following change fixes the issue:
```diff
--- a/ThirdParty/etl-20.40.0/include/etl/error_handler.h
+++ b/ThirdParty/etl-20.40.0/include/etl/error_handler.h
@@ -282,7 +282,7 @@ namespace etl
   /// Sets the assert function.
   /// The argument function signature is void(*)(const etl::exception&)
   //***************************************************************************
-  void set_assert_function(etl::private_error_handler::assert_function_ptr_t afptr)
+  inline void set_assert_function(etl::private_error_handler::assert_function_ptr_t afptr)
   {
     etl::private_error_handler::assert_handler<0>::assert_function_ptr = afptr;
   }
```

It seems to me this function needs to be marked either ```static``` or ```inline```, right? Because putting a function definition in a header file is going to lead to multiple definition errors if it gets included by more than one TU.

Or am I doing something else wrong?

jwellbelove, 2025-05-23:
You are correct, it should be `inline`.

jwellbelove, 2025-05-23:
Do you want to do a PR, or shall I handle it?

TheComet, 2025-05-23:
> Do you want to do a PR, or shall I handle it?

I'll make a PR

TheComet, 2025-05-23:
> Do you want to do a PR, or shall I handle it?

https://github.com/ETLCPP/etl/pull/1106