ETLCPP/etl #1118: Request: Constexpr constructible flat_map, map, and set containers.

drewr95, 2025-06-05:
I don't know how achievable this is, but it would be greatly useful on my embedded application to be able to place these data structures in ROM while also getting the algorithm performance benefit.  Currently if I want to create a basic map, I have to create a struct that has the key and the item it maps to (like a pair<>) and then put it into the array (O(N)).

Off the top of my head I think the challenge would be to find a place where an ASSERT could live that would make sure the keys are already in order.

jwellbelove, 2025-06-07:
Creating `constexpr` containers for the ETL is tricky.
The ETL uses placement-new to construct elements into uninitialised memory. This currently cannot be `constexpr`.

The C++ committee, in their infinite wisdom, decided that they would do what they have done with some other parts of C++ and give certain parts of the STL _special privileges_.

C++20 introduced `std::construct_at` that uses placement new internally; the compiler gives it a special pass to be `constexpr`.

Any code that excludes the use of the STL cannot write their own `constexpr construct_at` as the compiler will not recognise it for special treatment.

There is only one way, in C++20, that may work. That is to copy the signatrue used in the compiler's STL for `construct_at`, putting it into the `std` namespace.

This is a technique I had to use to support `std::initializer_list` in the ETL. That is another class that is closely coupled with the compiler. In the ETL there are implementations of `std::initializer_list` with _exactly_ the correct signature that the compilers will recognise.

This close coupling of the language and STL is a big issue for libraries like the ETL. It makes certain parts of the language next to impossible to use without the STL.

The only real way to get ETL `constexpr` containers, without the C++20 and the STL is to do something like you have described. It _may_ be possible to modify the _ext containers to use a `constexpr` external buffer, but I have a feeling that the `i*base_class` implementations may exclude this.

Other than only allowing `constexpr` ETL containers for C++20 + STL use only, the answer may be to create a set of  containers solely for `constexpr` use. They would undoubtedly be incompatible with the `i*base_class` classes.

drewr95, 2025-06-07:
Thank you for the explanation @jwellbelove as I did not realize that.  I think you can do the `i*base_class` approach provided there are no virtual methods.  I believe this is mostly the case, and since this library is intended for embedded use, I think it's safe to assume that deletion of the base class pointer should be avoided, thus the base class dtors can be made non-virtual and moved to `protected:`.

Other than that, yeah, I assumed there would be `const_*` equivalents of those containers.  For us unfortunate ones that are dealing with <C++20, I would be in high favor of it.  

I could try playing around with both options, I think `set` would be the easiest one to start with, off the top of my head.

jwellbelove, 2025-06-07:
Example:

```
// Using template deduction, constexpr vector of 10 int.
static constexpr etl::trivial_vector tv(0, 1, 2, 3, 3, 5, 6, 7, 8, 9);
```

drewr95, 2025-06-07:
Using the trivial verbiage seems like the right convention. I think requiring default constructible items is understandable.  A trivial_flat_map and trivial_set.

Curious though, what is the purpose of the trivial vector when arrays + spans exist?

jwellbelove, 2025-06-08:
I've abandoned `trivial_vector`, as, if it is just an implementation that creates a read only vector of fixed size, then its interface is no different from an `etl::array`.

`etl:array` is already `constexpr` compatible.

drewr95, 2025-06-12:
That is awesome!

jwellbelove, 2025-06-13:
There is another advantage to these `constexpr` containers.
As they will use contiguous storage, they could be accessed using `etl::span`.

```
using Node       = std::pair<int, int>;
using TrivialMap = trivial_map<Node, 10>;

constexpr TrivialMap tm{ Node{0, 1}, Node{1, 2}, Node{2, 3}, Node{3, 4}, Node{4, 5},
                         Node{5, 6}, Node{6, 7}, Node{7, 8}, Node{8, 9}, Node{9, 10} };

etl::span<Node> sp(tm.lower_bound(Node{2, 3}), tm.lower_bound(Node{7, 8}));

// sp contains the span of Node{2, 3}, Node{3, 4}, Node{4, 5}, Node{5, 6}, Node{6, 7}, Node{7, 8}
```

drewr95, 2025-06-13:
Oh that is fantastic!  I did a quick search and I'm surprised there is no `std::is_unique`.  You might want to create one and put it in `algorithm.h`?   Or maybe I'm not using the correct search terms.

jwellbelove, 2025-06-13:
I started a feature branch this morning and created the basics of `etl::const_map`. I'm going with that name at the moment as it will be designed for `const` and `constexpr` use only.
I'll give you the branch link when I think I have something for you to try out.

jwellbelove, 2025-06-16:
I have an implementation of `etl::const_map` in this branch.

`origin/feature/#1118-Request--Constexpr-constructible-flat_map,-map,-and-set-containers`

jwellbelove, 2025-07-16:
Updated 20.42.0

closed 2025-07-16 by jwellbelove