ETLCPP/etl #1126: to_arithmetic does not compile on C++98

HispidaBytes, 2025-06-13:
### Symptom
Compiling etl::to_arithmetic fails on C++98 compilers. I've tried this with a old IAR EWARM 7.80 and gcc 13.3.0 configured to use C++98.

This is the output of gcc 13.3.0 configured to use C++98:

```
[main] Building folder: /home/user/dev/etl_test/build 
[build] Starting build
[proc] Executing command: /usr/bin/cmake --build /home/user/dev/etl_test/build --config Debug --target all -j 18 --
[build] [ 50%] Building CXX object CMakeFiles/etl_test.dir/main.cpp.o
[build] In file included from /home/user/dev/etl_test/main.cpp:3:
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/to_arithmetic.h: In instantiation of ‘typename etl::enable_if<etl::is_integral<T>::value, etl::to_arithmetic_result<TValue> >::type etl::to_arithmetic(basic_string_view<TChar>, radix::value_type) [with TValue = int; TChar = char; typename enable_if<is_integral<T>::value, to_arithmetic_result<TValue> >::type = to_arithmetic_result<int>; radix::value_type = unsigned char]’:
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/to_arithmetic.h:855:45:   required from ‘typename etl::enable_if<etl::is_integral<T>::value, etl::to_arithmetic_result<TValue> >::type etl::to_arithmetic(const ibasic_string<TChar>&, radix::value_type) [with TValue = int; TChar = char; typename enable_if<is_integral<T>::value, to_arithmetic_result<TValue> >::type = to_arithmetic_result<int>; radix::value_type = unsigned char]’
[build] /home/user/dev/etl_test/main.cpp:12:68:   required from here
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/to_arithmetic.h:758:80: error: no matching function for call to ‘absolute_unsigned(const int&)’
[build]   758 |           const accumulator_type maximum = is_negative ? etl::absolute_unsigned(etl::integral_limits<TValue>::min)
[build]       |                                                          ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build] In file included from /home/user/dev/etl_test/build/_deps/etl-src/include/etl/gcd.h:35,
[build]                  from /home/user/dev/etl_test/build/_deps/etl-src/include/etl/algorithm.h:47,
[build]                  from /home/user/dev/etl_test/build/_deps/etl-src/include/etl/basic_string.h:35,
[build]                  from /home/user/dev/etl_test/build/_deps/etl-src/include/etl/string.h:35,
[build]                  from /home/user/dev/etl_test/main.cpp:2:
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/absolute.h:75:5: note: candidate: ‘template<class T, class TReturn> typename etl::enable_if<etl::is_signed<T>::value, TReturn>::type etl::absolute_unsigned(T)’
[build]    75 |     absolute_unsigned(T value) ETL_NOEXCEPT
[build]       |     ^~~~~~~~~~~~~~~~~
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/absolute.h:75:5: note:   template argument deduction/substitution failed:
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/to_arithmetic.h:758:80: note:   couldn’t deduce template parameter ‘TReturn’
[build]   758 |           const accumulator_type maximum = is_negative ? etl::absolute_unsigned(etl::integral_limits<TValue>::min)
[build]       |                                                          ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/absolute.h:89:5: note: candidate: ‘template<class T> typename etl::enable_if<etl::is_unsigned<T>::value, T>::type etl::absolute_unsigned(T)’
[build]    89 |     absolute_unsigned(T value) ETL_NOEXCEPT
[build]       |     ^~~~~~~~~~~~~~~~~
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/absolute.h:89:5: note:   template argument deduction/substitution failed:
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/absolute.h: In substitution of ‘template<class T> typename etl::enable_if<etl::is_unsigned<T>::value, T>::type etl::absolute_unsigned(T) [with T = int]’:
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/to_arithmetic.h:758:80:   required from ‘typename etl::enable_if<etl::is_integral<T>::value, etl::to_arithmetic_result<TValue> >::type etl::to_arithmetic(basic_string_view<TChar>, radix::value_type) [with TValue = int; TChar = char; typename enable_if<is_integral<T>::value, to_arithmetic_result<TValue> >::type = to_arithmetic_result<int>; radix::value_type = unsigned char]’
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/to_arithmetic.h:855:45:   required from ‘typename etl::enable_if<etl::is_integral<T>::value, etl::to_arithmetic_result<TValue> >::type etl::to_arithmetic(const ibasic_string<TChar>&, radix::value_type) [with TValue = int; TChar = char; typename enable_if<is_integral<T>::value, to_arithmetic_result<TValue> >::type = to_arithmetic_result<int>; radix::value_type = unsigned char]’
[build] /home/user/dev/etl_test/main.cpp:12:68:   required from here
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/absolute.h:89:5: error: no type named ‘type’ in ‘struct etl::enable_if<false, int>’
[build] In file included from /home/user/dev/etl_test/build/_deps/etl-src/include/etl/type_traits.h:58,
[build]                  from /home/user/dev/etl_test/build/_deps/etl-src/include/etl/algorithm.h:42:
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/expected.h: In instantiation of ‘etl::unexpected<TError>& etl::unexpected<TError>::operator=(const etl::unexpected<TError>&) [with TError = etl::to_arithmetic_status]’:
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/to_arithmetic.h:186:25:   required from ‘etl::to_arithmetic_result<TValue>& etl::to_arithmetic_result<TValue>::operator=(unexpected_type) [with TValue = int; unexpected_type = etl::unexpected<etl::to_arithmetic_status>]’
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/to_arithmetic.h:741:16:   required from ‘typename etl::enable_if<etl::is_integral<T>::value, etl::to_arithmetic_result<TValue> >::type etl::to_arithmetic(basic_string_view<TChar>, radix::value_type) [with TValue = int; TChar = char; typename enable_if<is_integral<T>::value, to_arithmetic_result<TValue> >::type = to_arithmetic_result<int>; radix::value_type = unsigned char]’
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/to_arithmetic.h:855:45:   required from ‘typename etl::enable_if<etl::is_integral<T>::value, etl::to_arithmetic_result<TValue> >::type etl::to_arithmetic(const ibasic_string<TChar>&, radix::value_type) [with TValue = int; TChar = char; typename enable_if<is_integral<T>::value, to_arithmetic_result<TValue> >::type = to_arithmetic_result<int>; radix::value_type = unsigned char]’
[build] /home/user/dev/etl_test/main.cpp:12:68:   required from here
[build] /home/user/dev/etl_test/build/_deps/etl-src/include/etl/expected.h:144:7: error: invalid application of ‘sizeof’ to incomplete type ‘ETL_ETL_STATIC_ASSERT_FAILED<false>’
[build]   144 |       ETL_STATIC_ASSERT(etl::is_copy_constructible<TError>::value, "Error not copy assignable");
[build]       |       ^~~~~~~~~~~~~~~~~
[build] gmake[2]: *** [CMakeFiles/etl_test.dir/build.make:76: CMakeFiles/etl_test.dir/main.cpp.o] Error 1
[build] gmake[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/etl_test.dir/all] Error 2
[build] gmake: *** [Makefile:91: all] Error 2
[proc] The command: /usr/bin/cmake --build /home/user/dev/etl_test/build --config Debug --target all -j 18 -- exited with code: 2
[driver] Build completed: 00:00:00.373
[build] Build finished with exit code 2

```

### How to reproduce

Use the following CMakeList.txt and main.cpp with gcc 13.3.0

**CMakeLists.txt**
```
cmake_minimum_required(VERSION 3.10.0)
project(etl_test VERSION 0.1.0 LANGUAGES C CXX)

set(CMAKE_CXX_STANDARD 98)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)  # Optional: disables GNU extensions like 

Include(FetchContent)

FetchContent_Declare(
  etl
  GIT_REPOSITORY https://github.com/ETLCPP/etl
  GIT_TAG        20.41.7
)

FetchContent_MakeAvailable(etl)

add_executable(etl_test main.cpp)
target_link_libraries(etl_test PRIVATE etl::etl)

```

**main.cpp**
```
#include <iostream>
#include "etl/string.h"
#include "etl/to_arithmetic.h"

int main(int, char**){
    std::cout << "Hello, from etl_test!\n";

    etl::string<100> test = "10";
    etl::to_arithmetic_result<int> parsed = etl::to_arithmetic<int>(test, 16);
    if(parsed.has_value())
    {
        std::cout << "parsed: " << parsed.value() << "\n";
    }
    else
    {
        std::cout << "Error parsing!\n";
    }
}

```

jwellbelove, 2025-06-20:
Can you try this patch?

[etl-#1126.patch](https://github.com/user-attachments/files/20842458/etl-.1126.patch)

jwellbelove, 2025-06-20:
It may fail with  [-Werror=unused-local-typedefs]

jwellbelove, 2025-06-21:
I have a new patch.

[etl-#1126.patch](https://github.com/user-attachments/files/20845614/etl-.1126.patch)

jwellbelove, 2025-06-23:
`ETL_STATIC_ASSERT(false, "No user defined specialisation of etl::underlying_type for this type");`
This line has been removed from the development branch.

Try this 

[etl-#1126-type_traits.patch](https://github.com/user-attachments/files/20869379/etl-.1126-type_traits.patch)

jwellbelove, 2025-06-26:
I think it may be failing due to the etl::enable_if for `etl::bit_cast`.
It would be useful to know what the input parameters were for `etl::to_arithmetic`.
What test was it executing at the time?

jwellbelove, 2025-06-26:
OK, I have another patch to try. This breaks the calculation of the result into two steps, with a calculation of the value, and then assignment to the result type.

[etl-#1126-to_arithmetic.patch](https://github.com/user-attachments/files/20923202/etl-.1126-to_arithmetic.patch)

jwellbelove, 2025-06-30:
I'm away on holiday in South Wales for the rest of the week, and I deliberately didn't bring the laptop so as not to be temped to work on the ETL whilst I was away 😄. I'll be back next Saturday evening.

HispidaBytes, 2025-07-01:
That was definitively a smart move 👍 
Enjoy your holiday 😊

jwellbelove, 2025-07-15:
I've tried this with GCC 12.3, and I get no compilation errors.

```
cmake_minimum_required(VERSION 3.10.0)
project(etl_test VERSION 0.1.0 LANGUAGES C CXX)

set(CMAKE_CXX_STANDARD 98)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)  # Optional: disables GNU extensions like 

Include(FetchContent)

FetchContent_Declare(
  etl
  GIT_REPOSITORY https://github.com/ETLCPP/etl
  GIT_TAG        20.41.7
)

FetchContent_MakeAvailable(etl)

add_executable(etl_test main.cpp)
target_link_libraries(etl_test PRIVATE etl::etl)
```

```
#include <iostream>
#include "etl/string.h"
#include "etl/to_arithmetic.h"
#include "etl/static_assert.h"

int main(int, char**){
  //std::cout << "Hello, from etl_test!\n";

  etl::string<100> test = "10";
  etl::to_arithmetic_result<int> parsed = etl::to_arithmetic<int>(test, 16);
  if(parsed.has_value())
  {
    std::cout << "parsed: " << parsed.value() << "\n";
  }
  else
  {
    std::cout << "Error parsing!\n";
  }
}
```

jwellbelove, 2025-07-15:
OK, I've managed to get it to fail now.

jwellbelove, 2025-07-20:
Fixed 20.42.2

closed 2025-07-20 by jwellbelove