ETLCPP/etl #1129: State change from on_enter_state is not processed by etl::hfsm

r2r0, 2025-06-17:
It looks that return value from on_enter_state method is not taken into account when event is processed in the etl::hfsm::receive - HFSM stays in the (just) entered state.
I have HFSM state where on_enter_state can conditionally request skipping to another state (in this case it is a parent state of just entered state).
Even if condition is true and on_enter_state returns value different than No_State_Change then HFSM stays in the same state.

ETL version 20.41.7

jwellbelove, 2025-06-18:
In the documentation it states:-
_Note:For an `etl::hfsm, on_enter_state()` cannot force a state change. It must return `etl::ifsm_state::No_State_Change`._

It also says:-
_If it does not, then an ETL_ASSERT will be raised._
It seems that this is not currently true.

The original creator of the HFSM (Jeremy Overesch) obviously had a reason for requiring this condition.
I don't know if this is resolvable or not.

r2r0, 2025-06-18:
I am sorry, I overlooked this note (I know it is on the top of page...)
It looks that it is somehow design assumption/limitation and such change may be not worth effort and potential bug introduction.
So, from my side will be ok to close this issue.