ETLCPP/etl #1153: ROM size increase after algorithm_exception introduction

r2r0, 2025-07-21:
Commit [f0af24b6038ba2801826db5974ac9baed980fece](https://github.com/ETLCPP/etl/commit/f0af24b6038ba2801826db5974ac9baed980fece) causes step growth of ROM usage by about 1.5k. Unfortunatey for me (my project) it is non negligible value and it is reason why I report this issue.
Further investigation lead me to the lines where there are added `algorithm_exception` and `algorithm_error` classes (and their usages).
Even if I undefine ETL_VERBOSE_ERRORS the increase of generated code size is still significant (a slightly above 1k).

Platform: Cortex-M3, GCC 12.2.0 (Zephyr toolchain 0.17.0 )

r2r0, 2025-07-21:
I use `ETL_LOG_ERRORS`.

If I comment out lines with `ETL_ASSERT` in the copy_s then size increase is only about 70 bytes.

BTW> Compilation with disabled exceptions and RTTI.

jwellbelove, 2025-07-21:
I can reduce it by 24 bytes by replacing the two ETL_ASSERT lines with one.

`ETL_ASSERT((s_size >= 0) || (d_size >= 0), ETL_ERROR(algorithm_error));`

jwellbelove, 2025-07-21:
It's interesting that our compilers are so different in how many bytes they use for the same code.

jwellbelove, 2025-07-22:
That makes sense.