ETLCPP/etl #1156: Name conflict between template parameter and status register macro on MSP430 devices

fleckerbr, 2025-07-24:
Numerous templates in etl include a parameter named `N` such as the `nth_base` struct in `type_traits.h`:
```cpp
  //***************************************************************************
  /// Get the Nth base of a recursively inherited type.
  /// Requires that the class has defined 'base_type'.
  //*************************************************************************** 
  // Recursive definition of the type.
  template <size_t N, typename TType>
  struct nth_base
  {
    typedef typename nth_base<N - 1U, typename TType::base_type>::type type;
  };
```

Unfortunately, TI has a status register bit macro in their MSP430 header files that shares the same name and prevents me from compiling certain etl features without performing modifications to etl's headers. For reference, `msp430fr5xx_6xxgeneric.h` defines the following:
```c
/************************************************************
* STATUS REGISTER BITS
************************************************************/

#define C                      (0x0001)
#define Z                      (0x0002)
#define N                      (0x0004)
#define V                      (0x0100)
#define GIE                    (0x0008)
#define CPUOFF                 (0x0010)
#define OSCOFF                 (0x0020)
#define SCG0                   (0x0040)
#define SCG1                   (0x0080)
```

Would it be possible to rename of all the template parameters named `N` to prevent this issue or is there perhaps a more elegant solution?

Thank you

jwellbelove, 2025-07-25:
A quick search of the codebase highlights 67 instances of `size_t N`.
Other single character template parameters that could potentially fall foul of macros include `size_t I` (13 instances), `typename T` (2281 instances), `typename U` (130 instances).

There are also many All-Caps template parameters that could possibly clash with macros.
```
size_t NBITS
size_t POSITION
size_t ID
```

There is an outstanding issue I raised a while ago to update the codebase to use Pascal Case, with underscores (`Pascal_Underscore_Case`), which is unlikely to clash with any macro.

jwellbelove, 2025-07-25:
Another option is to push/pop any clashing macro definitions.

fleckerbr, 2025-07-25:
Pushing and popping macro definitions sounds like the most foolproof way to get around this issue for compilers that support it. Since the push/pop pragmas weren't added to the C++ specification until C++11, I'm not sure if they meet etl's C++3 compatibility objectives though.

I guess that you could just rename all the parameters and also push/pop macro definitions that would clash with the new parameter names on compilers that support it. That would "fix" the issue while also providing a bit more fool-proofing for compilers that support the push/pop pragmas.

Admittedly, I'm quite new to etl and C++ as a whole so I'm not particularly confident in saying that's the best solution available.

jwellbelove, 2025-07-27:
I've decided to go with a general renaming of things that may clash with a macro, without breaking any of the public API.

closed 2025-08-19 by jwellbelove