ETLCPP/etl #1157: Build error with non-virtual destructor

KammutierSpule, 2025-07-25:
`deleting object of polymorphic class type 'etl::callback_timer_locked<254>' which has non-virtual destructor might cause undefined behavior [-Werror=delete-non-virtual-dtor]`

I'm building with -Wall -Wextra -Wpedantic -Werror

From what I understand, it needs on interface classes a `virtual ~Class() = default;` to fix it.

Is this expected or is there other suggestion for this?

jwellbelove, 2025-07-26:
The ETL's `etl::callback_timer_locked`, like many types in the ETL, is not intended to be destructed via a pointer to the 'implementation' base class.

Many base classes in the ETL are not polymorphic as they are not 'interface' classes, they are the size-independent 'implementation' classes that can be used to access many different derived types via a pointer of reference. This cuts code bloat when many differently sized instantiations of a template are created.

Example:
```
etl::callback_timer_locked<1>   timer1;
etl::callback_timer_locked<10>  timer10;
etl::callback_timer_locked<254> timer254;
```

All of the above use the same `etl::icallback_timer` implementation.

jwellbelove, 2025-07-26:
Adding a virtual destructor would just add an unnecessary vtable to the objects.

closed 2025-09-21 by jwellbelove