ETLCPP/etl #1159: etl::fsm helper to check states id sequence at compile time

puzrin, 2025-07-27:
Currently etl::fsm check state id-s at runtime. I did a small helper to do that at compile time.

``` cpp
#pragma once

#include <etl/fsm.h>
#include <etl/tuple.h>
#include <etl/type_traits.h>

namespace etl_ext {

namespace detail {

// Compile-time: S::ID must equal its index in the type list (0..N-1)
template <size_t N, typename...> struct check_ids : etl::true_type {};

template <size_t N, typename S0, typename... Rest>
struct check_ids<N, S0, Rest...>
  : etl::integral_constant<bool,
        (S0::STATE_ID == N) && detail::check_ids<N + 1, Rest...>::value> {};

} // namespace detail

template <typename... FsmStates>
struct fsm_state_pack {
    static_assert(detail::check_ids<0, FsmStates...>::value,
        "State IDs must be 0..N-1 and in order");
    static_assert(sizeof...(FsmStates) > 0, "At least one state is required");

    etl::tuple<FsmStates...> storage{};
    etl::ifsm_state* states[sizeof...(FsmStates)]{ &etl::get<FsmStates>(storage)... };

    static constexpr size_t size = sizeof...(FsmStates);

    template <typename S>
    S& get() { return etl::get<S>(storage); }

    template <typename S>
    const S& get() const { return etl::get<S>(storage); }
};

} // namespace etl_ext
```

Usage:

```cpp
etl_ext::fsm_state_pack<S1, S2> myfsm_state_list;

myfsm.set_states(myfsm_state_list.states, myfsm_state_list.size);
```

Can it be usefult in mainstream?

jwellbelove, 2025-08-03:
An overload of `set_states` could probably be added to take a `etl_ext::fsm_state_pack` directly, and then the original `set_states` could be used for adding states in the 'old' way.

puzrin, 2025-08-04:
If states are usually created with a default constructor, I'd suggest promoting something like state-pack as a default in examples. Current manual creation of every state, then writing an array of instances, looks boring and a bit out of sync with the beauty of templates.

But I don't know all possible ways of states usage, and my personal opinion is not qualified enough.

puzrin, 2025-08-04:
Personally, I'm thinking about creating an alternative to etl::fsm with fully static states. Something like tinyfsm, but with the convenience of `etl::fsm` with state switches via `return XX`. Existing `etl::fsm` eats some RAM for each state instance, and I expect complaints about unnecessary expenses in my USB PD Sink library.

May be, it will be enough to keep `etl::fsm` with light polish, without breaking changes, and apply more aggressive optimisations in the new FSM.

jwellbelove, 2025-08-19:
Released 20.43.0