ETLCPP/etl #1160: Feature request: rounded division (similar to scaled rounding)

positron96, 2025-07-28:
I find rounded integer division to be as useful (if not more) as scaled rounding, which is already in ETL. 

Proposal is to add a set of functions that calculate `a/b` in integers with different rounding options. As far as I understand, the code is already there inside scaled rounding, it just needs to be refactored out, and divisor needs to be an ordinary argument, not template argument.

jwellbelove, 2025-08-09:
I've looked at this, and created a prototype implementation. The code in 'scaled rounding' was not really applicable to the 'unscaled rounding' involved in this functionality.

What needs to be resolved is how to handle the condition of when the denominator is zero.

1: Make it the responsibility of the caller not to pass a denominator of zero.

2: Raise an `ETL_ASSERT`.

3: Return an `etl::optional`.

At the moment, my implementation is `constexpr` for C++14 and above.

The same may not be true for (2).

It may be `constexpr` for (3) if I have specialised `etl::optional` for fundamental or trivial types. (I'll need to check).

positron96, 2025-08-09:
For me, even 1st case would be ok. Regarding ETL_ASSERTs, what would happen when they are disabled (which is default as far as I know)?

positron96, 2025-08-19:
Thank you for amazing library!