ETLCPP/etl #1162: etl::fsm, allow set_state(new_id) from the outer calls.

puzrin, 2025-08-04:
Sometimes it's needed to change the FSM state from the outer code. Currently, I have to add `MsgTransitTo{id}` event to every state, and run `receive(MsgTransitTo{id})`. TBH, that works but looks ugly.

IMHO, it would be nice to have a public `.set_state(id)` method. It can be "blocked" during `receive()` execution, which looks reasonably safe.

puzrin, 2025-08-04:
I think, problem is specific for etl::fsm. Because it's approach it's to define new state via `return XXX` (and that's really cool).

Thechnically, if problem is specific for `etl::fsm`, extra public method of state change should be also specific for `etl::fsm`

AFAIK, HFSM uses direct calls to change state, so there should be no intersection at all (problem not exists there). If under the hood HFSM is inherited from FSM, then such extended method should be deleted for safety (because it is etl::fsm specific). Or chain of inheritance should be readjusted (BaseFSM => {FSM, HFSM})

jwellbelove, 2025-08-05:
`set_state` will have to do a lot of what receive does when changing states, such as `on_exit_state`/`on_enter_state` etc.
If I extract the state change processing to a `private` `virtual`, then the HFSM override can do the right thing if the HFSM is accessed from an `etl::fsm*`.

This actually makes the `receive` override in `etl::hfsm` redundant.

jwellbelove, 2025-08-05:
On the other hand, should this forced state change ignore the effects of `on_exit_state`/`on_enter_state` and just do a hard direct jump to the new state?

puzrin, 2025-08-05:
I mean, there are 2 abstraction layers:

- etl::fsm architecture (global)
- the concrete implementation of HFSM/FSM

At architecture level, there are no logical collision, set_state() of etl::fsm is specific thing for edge cases. This is not related to HFSM anyhow.

At implementation level, if HFSM inherited from FSM - collision can exist. But this is implementation issue, not principal problem of FSM design theory. I'm not sure this method can be reused in HFSM - this can conflict with nature of etl::fsm.

> On the other hand, should this forced state change ignore the effects of on_exit_state/on_enter_state and just do a hard direct jump to the new state?

When demanded set_state(), I meaned it should execute exit/enter machinery to be useful.

puzrin, 2025-08-05:
> The only difference being that it is a direct action and ignores the effect (such as queueing messages) of an overridden `receive()` in the derived class.

I think, this should not be a problem.

- .set_state() method is a companion to `return XXX`, mutually exclusive, allowed to be called only out of states (set/clear flag on `.receive()` begin/end and decline `.set_state()` execution if flag exists).
- Users of this feature are expected to understand very well what they do. FSM reaction is sync. Queued input is the upper abstraction layer. If someone wishes to embed an input queue and fire `.set_state()` - that's a design mistake, not a problem with `.set_state()`.

jwellbelove, 2025-08-05:
I think that `set_state` should have another name, as it too similar to `set_states`, which does a completely different thing.

`force_state`
`goto_state`
`select_state`
`change_state`
`transition_to`  <<< I quite like this one
`switch_state`