ETLCPP/etl #1166: Feature Request: implement not_null<T*>

drewr95, 2025-08-14:
I don't know if it was ever requested here but I think it would be useful to have for explicit data that can't be null that doesn't use `std::reference_wrapper<>`.

For instance, when I have classes that get dependency injected references to other modules, I make the constructor arg take the reference & to the object so it can't be null, but my underlying data member is a pointer because the clang-tidy warns about the class being non-copyable.  I don't necessarily want to have internal `std::reference_wrapper<>` member variables to just call get() any time I want to access it.  

Thoughts? I don't know how many others would want this but I think there is a difference (however small) between `std::reference_wrapper<>` and `not_null<>`

jwellbelove, 2025-08-15:
Can you try this initial implementation?

[etl-#1166-no_null-2025-08-15.patch](https://github.com/user-attachments/files/21799334/etl-.1166-no_null-2025-08-15.patch)

drewr95, 2025-08-16:
Thanks so much! I'll give this a shot.  A couple things when I was reviewing:

1.  It would be nice to have all the methods marked as `noexcept`.
2. It might be nice to see what methods can be `constexpr` and what cannot be.  Something I was thinking about is it might be nice to stoire a `not_null` in ROM.

jwellbelove, 2025-08-17:
I think that for `etl::not_null<etl::unique_ptr<T, TDeleter>>` moves should be deleted, as moving will break the contract that `etl::not_null` will not be null.

jwellbelove, 2025-08-17:
I've added `constexpr` and `noexcept`, and removed the access to the underlying type.
Also disabled 'move' from `not_null<etl::unique_ptr<T, TDeleter>>` operations.

drewr95, 2025-08-17:
I just tried your patch and applying the patch does not work for me:

> root@bec145203f14:/dependencies/thirdparty/etl# git apply etl-.1166-no_null-2025-08-15.patch 
> etl-.1166-no_null-2025-08-15.patch:81: trailing whitespace.
>     not_null& operator =(const etl::not_null<T*>& rhs) 
> etl-.1166-no_null-2025-08-15.patch:94: trailing whitespace.
>     not_null& operator =(T* rhs) 
> etl-.1166-no_null-2025-08-15.patch:134: trailing whitespace.
>     T* operator->() const 
> etl-.1166-no_null-2025-08-15.patch:166: trailing whitespace.
>       : u_ptr(etl::move(u_ptr_)) 
> etl-.1166-no_null-2025-08-15.patch:178: trailing whitespace.
>       : u_ptr(etl::move(other.u_ptr)) 
> error: include/etl/not_null.h: No such file or directory
> error: patch failed: test/vs2022/etl.vcxproj.filters:3701
> error: test/vs2022/etl.vcxproj.filters: patch does not apply

jwellbelove, 2025-08-17:
In the user's code you would do something like this (heap memory example).

```
using u_ptr_t = etl::unique_ptr<int>;

// u_ptr points to an int on the heap containing the value 123.
u_ptr_t u_ptr(new int{ 123 });

// Moves the u_ptr to etl::not_null<u_ptr_t.
etl::not_null<u_ptr_t> nn(etl::move(u_ptr));
```

If you were to pass an uninitialised `etl::unique_ptr` or one that has been assigned a null pointer, then `etl::not_null` would `ETL_ASSERT`.

drewr95, 2025-08-17:
I don't use heap.  But thank you! I'll try to get the branch.  When I used CMake Fetch, it was having issues getting the branch: `feature/https://github.com/ETLCPP/etl/issues/1166-Feature-Request--implement-not_null`

jwellbelove, 2025-08-17:
> I don't use heap.

`unique_ptr` doesn't have to be used with the heap. The second template parameter allows you to define a 'deleter' that returns the storage back to where it was allocated from. By default it is the standard `delete`, but it could be anything. You could use `unique_ptr` with a custom fixed capacity allocator, and have the deleter call the allocation's 'release' function.

jwellbelove, 2025-08-17:
> When I used CMake Fetch, it was having issues getting the branch:

You could always download the [raw file](https://github.com/ETLCPP/etl/blob/feature/%231166-Feature-Request--implement-not_null/include/etl/not_null.h) if you're having issues.