ETLCPP/etl #1167: What happened to compiler built-ins? unresolved external symbol memmove

igromanru, 2025-08-21:
Hello, I'm trying to evaluate whether the library is something I can use in my projects, which don't use CRT or STL at all.  
I've created this `etl_profile.h`:
```cpp
#ifndef EMPLE_H
#define EMPLE_H

#define ETL_COMPILER_GENERIC
#define ETL_CPP11_SUPPORTED 1
#define ETL_CPP14_SUPPORTED 1
#define ETL_CPP17_SUPPORTED 1
#define ETL_NO_NULLPTR_SUPPORT 0
#define ETL_NO_LARGE_CHAR_SUPPORT 0
#define ETL_CPP11_TYPE_TRAITS_IS_TRIVIAL_SUPPORTED 0
#define ETL_NO_STL

#endif
```
And now I'm just trying to see if I can use `etl::string` and `etl::wstring`.  

I’m on Windows with MSVC, using C++20. In my DLL project, I’ve set the following linker configuration:
 `/NODEFAULTLIB[libcmt.lib,libucrt.lib,libvcruntime.lib,libcmtd.lib,ucrt.lib,vcruntime.lib]`  

I get error: `error LNK2001: unresolved external symbol memmove`
In the documentation, I found [Enabling compiler built-ins](https://www.etlcpp.com/compiler-built-ins.html), but I don’t see the suggested defines anywhere in the ETL code.
Defining `ETL_USE_BUILTIN_MEMMOVE 1` or `ETL_USING_MEM_BUILTINS` doesn’t change anything, which is expected since they don’t appear in the code at all.  
Are compiler built-ins gone now, or I'm just doing something wrong?
The project is so big and has so many possible Macro settings that even though I'm trying to read the documentation, I'm still confused.

jwellbelove, 2025-08-21:
I've started to modify the code to have a base `mem_copy`/`mem_move` that selectively calls `memcpy`/`memmove` or `__builtin_memcpy`/`__builtin_memmove`, dependant on configuration. The other templated `mem_copy`/`mem_move` functions will call these.

igromanru, 2025-08-21:
Alright, but do you have an idea what I'm doing wrong?  
<details>
<summary>Full compiler error</summary>

```
  1>dllmain.cpp
  1(indlude path)\etl\platform.h(437,25): error C2628: 'uint_least8_t' followed by 'char8_t' is illegal (did you forget a ';'?)
  1>(compiling source file 'src/dllmain.cpp')
  1(indlude path)\etl\platform.h(437,32): warning C4091: 'typedef ': ignored on left of 'unsigned char' when no variable is declared
  1>(compiling source file 'src/dllmain.cpp')
  1(indlude path)\etl\char_traits.h(119,31): error C2504: 'etl::char_traits_types<T>': base class undefined
  1(indlude path)\etl\char_traits.h(119,31): error C2504:         with
  1(indlude path)\etl\char_traits.h(119,31): error C2504:         [
  1(indlude path)\etl\char_traits.h(119,31): error C2504:             T=char8_t
  1(indlude path)\etl\char_traits.h(119,31): error C2504:         ]
  1>(compiling source file 'src/dllmain.cpp')
  1>    D:\Git\UnrealEngine\D7-SDKTest\D7-SDKTest\etl\include\etl\char_traits.h(119,31):
  1>    the template instantiation context (the oldest one first) is
  1>        D:\Git\UnrealEngine\D7-SDKTest\D7-SDKTest\etl\include\etl\string_view.h(898,1):
  1>        see reference to class template instantiation 'etl::char_traits<T>' being compiled
  1>        with
  1>        [
  1>            T=char8_t
  1>        ]
  1(indlude path)\etl\char_traits.h(121,44): error C2027: use of undefined type 'etl::char_traits_types<T>'
  1(indlude path)\etl\char_traits.h(121,44): error C2027:         with
  1(indlude path)\etl\char_traits.h(121,44): error C2027:         [
  1(indlude path)\etl\char_traits.h(121,44): error C2027:             T=char8_t
  1(indlude path)\etl\char_traits.h(121,44): error C2027:         ]
  1>(compiling source file 'src/dllmain.cpp')
  1>    D:\Git\UnrealEngine\D7-SDKTest\D7-SDKTest\etl\include\etl\char_traits.h(119,31):
  1>    see declaration of 'etl::char_traits_types<T>'
  1>        with
  1>        [
  1>            T=char8_t
  1>        ]
  1(indlude path)\etl\char_traits.h(121,55): error C3646: 'char_type': unknown override specifier
  1>(compiling source file 'src/dllmain.cpp')
  1(indlude path)\etl\char_traits.h(121,1): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
  1>(compiling source file 'src/dllmain.cpp')
  1(indlude path)\etl\char_traits.h(122,44): error C2027: use of undefined type 'etl::char_traits_types<T>'
  1(indlude path)\etl\char_traits.h(122,44): error C2027:         with
  1(indlude path)\etl\char_traits.h(122,44): error C2027:         [
  1(indlude path)\etl\char_traits.h(122,44): error C2027:             T=char8_t
  1(indlude path)\etl\char_traits.h(122,44): error C2027:         ]
  1>(compiling source file 'src/dllmain.cpp')
  1>    D:\Git\UnrealEngine\D7-SDKTest\D7-SDKTest\etl\include\etl\char_traits.h(119,31):
  1>    see declaration of 'etl::char_traits_types<T>'
  1>        with
  1>        [
  1>            T=char8_t
  1>        ]
  1(indlude path)\etl\char_traits.h(122,55): error C3646: 'int_type': unknown override specifier
  1>(compiling source file 'src/dllmain.cpp')
  1(indlude path)\etl\char_traits.h(122,1): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
  1>(compiling source file 'src/dllmain.cpp')
  1(indlude path)\etl\char_traits.h(123,44): error C2027: use of undefined type 'etl::char_traits_types<T>'
  1(indlude path)\etl\char_traits.h(123,44): error C2027:         with
  1(indlude path)\etl\char_traits.h(123,44): error C2027:         [
  1(indlude path)\etl\char_traits.h(123,44): error C2027:             T=char8_t
  1(indlude path)\etl\char_traits.h(123,44): error C2027:         ]
  1>(compiling source file 'src/dllmain.cpp')
  1>    D:\Git\UnrealEngine\D7-SDKTest\D7-SDKTest\etl\include\etl\char_traits.h(119,31):
  1>    see declaration of 'etl::char_traits_types<T>'
  1>        with
  1>        [
  1>            T=char8_t
  1>        ]
  1(indlude path)\etl\char_traits.h(123,55): error C3646: 'off_type': unknown override specifier
  1>(compiling source file 'src/dllmain.cpp')
[... 36 lines cut ...]

</details>

All I do is call the `EtlTest::Run();` function in the DllMain. 
```cpp
#include "etl/string.h"
#include "etl/wstring.h"

namespace EtlTest
{
	inline void Run()
	{
		etl::string<1024> normalString("Obi-Wan Kenobi");
		normalString += " 1337";

		etl::wstring<1024> wideString(L"Luke Skywalker");
		wideString += L" ";
		wideString += 8888;
	}
}
```
I've changed my project settings a bit to keep `/NODEFAULTLIB` but allow few CRT libs.  
No exceptions, `/MT` and `/GS-`. Nothing really special.  
[image]
[image]

jwellbelove, 2025-08-21:
The first error is puzzling as it doesn't like the `typedef`.
Maybe `uint_least8_t` is not defined?

jwellbelove, 2025-08-22:
Try adding `#define ETL_CPP20_SUPPORTED 1` to `etl_profile.h`

For a few years now it has been possible to do without defining the `ETL_CPPxx_SUPPORTED` macros, as `platform.h` should be able to automatically detect the C++ standard that is being used by the compiler.

For MSVC you need to use the flag `/Zc:__cplusplus`.

[image]